RaumZeit/LinuxOnAcerCB5-311 #1: Only reboot, screen flickers, not possible to log

bmalnersic, 2015-10-10:
Hi,

thanks for the great guide (https://www.tbi.univie.ac.at/~ronny/acer-cb5-311.html). I have been trying to install Arch Linux using the guide. However, after the CB is restarted, the Linux kernel boots and after few seconds, screen starts to flicker (like switching on and off). There is activity (kernel outputs lines) however I cannot login. If I change terminals, I can sometimes briefly see login prompt but is instantly overwritten by kernel logging and I am not able to login. It seems like there would be switching between X and console? I have left CB in such state for 15 minutes but nothing changed.
If I press OFF button, the flickering will stop, kernel will print logs related to system shutdown and will panic at the end.
What should I do?

Best regards,

   Blaz

RaumZeit, 2015-10-12:
Hi,

the massive kernel log spam should only 'overwrite' the tty login prompt. If you hit the enter key, you should see it again, and you should be able to login. Usually, login is also possible 'blindly', meaning, you do just enter the username, hit the enter-key, and enter the password, followed by enter-key again. However, this console switching seems a little bit odd...
Maybe, the latest armv7h image from archlinuxarm got updated, and lightdm refuses to start with the ChromeOS kernel. Another explanation may be that the ChromeOS kernel you are using is more recent (or maybe older), compared to the one I've been using...
I'll have a look at that issue as soon as I find some spare time.
I assume, you experience this problem right after rebooting from ChromeOS. At this stage, some problems are expected, since the ChromeOS kernel actually needs to be rebuoild to work properly with systemd in ArchLinux. I'll see whether I can add the installation of a proper kernel into the archlinux.sh install script.

Best,
Ronny

bmalnersic, 2015-10-12:
Hi,

thanks for replaying so quickly:) I did some more troubleshooting and figured out next things:
- Problem was, that X11 would not start and runevel is by default set to 5. I changed the kernel parameters to start in runlevel 3 and was able to login to console.
- xinitrc is by default configured to start twm with xclock and xterm. These apps were not present. First option was to install additional packages (xorg-twm xorg-xclock xterm xorg-xinit xorg-utils) which helped. I then rather commented out twm and added "exec startxfce4". XFCE4 worked fine (startx). 
- I am now trying to figure things with lightdm. I moved xinitrc to xprofile and started lightdm with sudo systemctl start lightdm successfully. It seems to work fine and I could try with runlevel 5.
- Kernel 3.10.18 was successfully compiled (I had to fix 0001-mwiflex-do-not-create-AP-and-P2P-interfaces-upon-dri.patch since content of main.c has changed. I installed new kernel and booted OK.
- Problems with USB 3.0: In order to easily switch between Chrome OS and ALArm I have installer ALArm on USB 2.0 drive. No issues there. However, when I moved to USB 3.0 drive, issues started once root partition started booting (got a lot of device not responding to set address errors, does not boot). There were no issues with USB 3.0 drive under Chrome OS. Maybe I should recompile kernel with some options? USB 2.0 is quite slow. Since not ALArm seems quite useful, I should probably copy it to eMMC. I guess that means I loose easy system switching... (CTRL+D, CTRL+U).
- Problem with Chromium browser; will not start. An error NvRmMemInit in PostForkChild is reported, Segfaults...
- I had to reinstall GTK3 since many apps (e.g. LibreOffice reported gtk3 related error upon startup). Reinstall helped...
- GLXgears: ~1150 FPS
- Problems with desktop resolution: I am not able to change resolution under Display. Fixed to 1920x1080.
- Problems with PPTP VPN; sometimes it gets established but fails after few seconds. I know, I know, IPsec would be much better :)
- Problems in Chrome OS: cannot enable offline mode for Drive, Docs.... Maybe settings of my gmail account? (company managed)?

That's it for now I guess. I have to follow the rest of your instructions through. Many thanks!!!

Best regards,

    Blaz

RaumZeit, 2015-10-25:
Hi Blaz,

I just updated the install script to address some of the issues you mentioned. All users should now by default start an xfce4 session due to the included .xinitrc
Additionally, I added the missing packages for the twm session as well, just in case.
I also updated the linux-nyan kernel package with a proper mwifiex patch. Compilation works flawless on my system.
About the other things you mentioned:
- USB 3.0 problem can be related to many things. I did not try to install on USB 3.0 device yet, but kernel support should be included. Try
```
dmesg | grep xhci
```
to find out whether your kernel supports it as well, or if some problems appear.

- The chromium browser problem is related to alarm's switch to GCC 5.2 a couple of months ago. The Acer chromebook is not the only device affected by that, nor is chromium the only program crashing. The root of the problems is currently being investigated by the alarm developers. Until there is a fix available, please use an older chromium browser package compiled with GCC 5.1, e.g. the one linked here: http://archlinuxarm.org/forum/viewtopic.php?f=60&t=9109&start=10#p48213

- The problems you experience in ChromeOS might be related to 'developer mode'. Have you found a fix for that?

Best,

Ronny

bmalnersic, 2015-10-28:
Hi Ronny,

thanks for an update. Here is where I stand now:
- Chromium browser works (found the same link:). Added to pacman ignore list.
- found your post regarding Thunderbird and Lightning, works.
- PPTP VPN works (my bad, I was on a same local subnet as network I was connecting to...).
- USB 3.0 works once system is booted (did not work for booting the root system).
- Installed ExaGear Desktop. Works great - using for Skype and Wine. Still have to figure out proper settings to get microphone working.

Remaining issues:
- Sometimes touchpad and USB mouse are not working when X starts. Usually it helps to wait for 1-3 minutes. Also when this happens, sound is not working properly (device NVIDIA Tegra Venice2 not detected). Kernel crash when loading module mousedev - Unable to handle kernel NULL pointer dereference at virtual address 00000000. Internal error: Oops: 207 (#1) SMP ARM. Modules linked in: mousedev(+) snd_hda_codec snd_soc_max98090 mwifiex_sdio mwifiex btmrvl_sdio cfg80211 bluetooth ip_tables
  - modules snd_soc_max98090 and snd_soc_tegra30_i2s are not used, module joydev does not get loaded 
  - I guess this is related to XFCE Mixer panel not having proper sound card set
  - **this happens often and is quite bothering. Are you also experiencing this issue?**
- Internal Microphone is not working. Tried with Sound Recorder and Skype.
- Have to figure out some additional keyboard mapping for Slovenian layout. Slovenian layout uses right Alt (Alt Gr) combinations with some keys like Alt Gr + V = @, Alt Gr + Q = \, Alr Gr + W = |...

Best regards.

   Blaz

bmalnersic, 2015-10-30:
Hi Ronny,

maybe we should open another post with this issues. Hopefully this will also help other users since I must admit it is really impressive how useful and fast this CB5-311 with Linux is...

Best regards,
   
     Blaz

closed 2015-10-30 by bmalnersic