RaumZeit/LinuxOnAcerCB5-311 #2: Occasionally touchpad is not working

bmalnersic, 2015-10-30:
Hi Ronny, 

Sometimes touchpad and USB mouse are not working when X starts. Usually it helps to wait for 1-3 minutes. Also when this happens, sound is not working properly (device NVIDIA Tegra Venice2 not detected). Kernel crash when loading module mousedev - Unable to handle kernel NULL pointer dereference at virtual address 00000000. Internal error: Oops: 207 (#1) SMP ARM. Modules linked in: mousedev(+) snd_hda_codec snd_soc_max98090 mwifiex_sdio mwifiex btmrvl_sdio cfg80211 bluetooth ip_tables
modules snd_soc_max98090 and snd_soc_tegra30_i2s are not used, module joydev does not get loaded
I guess this is related to XFCE Mixer panel not having proper sound card set
this happens often and is quite bothering. Are you also experiencing this issue?

Maybe it would be best if you could provide me with your custom kernel build?

Best regards,

   Blaz

RaumZeit, 2015-11-07:
This is interesting. I've never had any issue with input devices such as touchpad or mouse... Maybe this is related to your audio problems? I'll see that I put my compiled kernel package somewhere for you to test if this might solve any of your problems

bmalnersic, 2015-11-17:
You are correct, it is related to audio problems. I noticed that mouse will usualy start working after few minutes (but audio will not). If I try to unload and reload audio module in such keys, kernel will crash.
It would be great if you can upload you kernel.
(I will also try to recompile it again on my chromebook, maybe it will help).

RaumZeit, 2015-12-10:
Can you please install my latest core/linux-nyan kernel package (master branch in my PKGBUILDs repo) and check whether you still experience these problems? As I said earlier, I never encountered such a problem before.
BTW: official Arch Linux ARM repositories updated to xorg-server version 1.18 which is not compatible with the latest nvidia drivers. To keep a working TegraK1 GPU accelerated xorg, I had to add the following lines to /etc/pacman.conf
```
IgnoreGroup = xorg
IgnorePkg = xf86-input-evdev xf86-input-synaptics
```
to ignore any update to xorg, and the input drivers. This kept the old xorg-server 1.17

RaumZeit, 2015-12-11:
Bad news, I had this exact problem the first time yesterday, twice. After rebooting a second time, the sound device was properly discovered again, and everything worked as expected. This is strange, especially since it helped to reboot a couple of times. Maybe it's a hardware issue? I'll look into that!

RaumZeit, 2016-08-12:
This issue seems to be fixed for the last couple of kernel updates. I close this issue, please re-open if it persists for you with latest linux-nyan kernel package

closed 2016-08-12 by RaumZeit