RaumZeit/LinuxOnAcerCB5-311 #3: Occasionally sound does not work

bmalnersic, 2015-10-30:
Hi Ronny,

Occasinally once system boots, sound is not working properly (device NVIDIA Tegra Venice2 not detected). Kernel crash when loading module mousedev - Unable to handle kernel NULL pointer dereference at virtual address 00000000. Internal error: Oops: 207 (#1) SMP ARM. Modules linked in: mousedev(+) snd_hda_codec snd_soc_max98090 mwifiex_sdio mwifiex btmrvl_sdio cfg80211 bluetooth ip_tables
modules snd_soc_max98090 and snd_soc_tegra30_i2s are not used, module joydev does not get loaded
I guess this is related to XFCE Mixer panel not having proper sound card set
this happens often and is quite bothering. Are you also experiencing this issue?
Also when this happens touchpad and USB mouse are not working when X starts. Usually it helps to wait for 1-3 minutes.

Best regards,

    Blaz

RaumZeit, 2015-11-07:
Are you using alsa directly, or pulseaudio on top? I am asking this, because I can not reproduce your issue, and have no problems at all using pusleaudio. Btw. I use xfce4-volumed-pulse from AUR to control volume levels with keyboard keys.

RaumZeit, 2016-08-12:
This issue seems to be fixed with last couple of recent linux-nyan kernel package. Please re-open if it persists for you...

closed 2016-08-12 by RaumZeit