RaumZeit/LinuxOnAcerCB5-311 #4: Microphone does not work

bmalnersic, 2015-10-30:
Hi Ronny,

Internal Microphone is not working. Tried with Sound Recorder and Skype. I tried fiddling around with alsamixer, pamixer, ponymix, amixer. Based on what is written here maybe internal microphone would require additional kernel patching: [patch ASoC: tegra: add tegra+MAX98090 machine driver](http://www.mailbrowse.com/devicetree/54555.html)?

Best regards,

    Blaz

RaumZeit, 2015-11-07:
Microphone is not working for me either. I'll have a look at this patch and report back if this helps to bring the microphone back.

RaumZeit, 2015-12-10:
Microphone is working for me now. I added a package alarm/alsa-ucm-nyan in my master branch of the Arch Linux ARM PKGBUILDs fork. The ucm config files that are installed with that package help pulseaudio to determine which settings need to be applied for microphone usage. I just tested the microphone and camera by recording with vlc, everything worked fine.

closed 2015-12-10 by RaumZeit