RaumZeit/LinuxOnAcerCB5-311 #5: Coredumps (coredumpctl list)

bmalnersic, 2015-10-30:
Hi Ronny, 

I noticed that there are some coredumps happening in the background. My system is setup to start in console mode so no errors while in console. Once I start the X session (systemctl start lightdm) with XFCE, I get some issues:
- with /usr/bin/xfsettingsd
   - if I kill the daemon, various settings applied to XFCE (like theme) vanishes
      - if I then run the deamon in foreground (XFSETTINGSD_DEBUG=1 xfsettingsd --replace --no-daemon) settings are reapplied, but application crashes with SIGTRAP (ps still shows that app is  running in background).
- with /usr/lib/upower/upowerd
   - does not seem to be anything wrong

Seems this is not affecting usability and stability of the system. So I guess this is just a low importance note.

Best regards,
    
   Blaz

RaumZeit, 2015-11-07:
Hi Blaz,

I am also experiencing some coredumps in the background. As far as I can tell, they result from bugs in upowerd. I tried several times to dig into this, to find out what the actual problem might be. My guess is that upower tries to obtain information from dbus for a device that does not exist, maybe the battery or something else. As far as I know more, I'll let you know

RaumZeit, 2015-12-09:
The issue with upower coredumping all the time is fixed. You need to apply a small patch to the upower sources, see https://bugs.freedesktop.org/show_bug.cgi?id=93095
I already made a PKGBUILD for a modified upower version that includes that patch, which is available as extra/upower in the master branch of my PKGBUILDs fork. Thus, I close this issue.

closed 2015-12-09 by RaumZeit