RaumZeit/LinuxOnAcerCB5-311 #7: gpu-nvidia-tegra-k1 not supporting ABI 20

zette, 2016-04-04:
Since recent X.Org update, X server ABI was updated to version 20. It is not supported by current (21.4) Nvidia driver for TK1.

Arial7, 2016-04-08:
This is not an issue of this repo, NVIDIA just does not support the newer Xorg.

zette, 2016-04-08:
True, but PKGBUILD for should install or at least have as dependency not the latest X.Org but the one that is supported but packaged nvidia driver..

RaumZeit, 2016-04-09:
You are both right. I'll include the dependency for xorg < 1.18. However, this might not help at all for fresh install of ArchLinuxARM, since the older xorg package is not available anymore. Unfortunately, there has been no progress on NVIDIAs side to support ABI 20 (https://devtalk.nvidia.com/default/topic/901558/jetson-tk1/t4l-xorg-driver-and-xorg-1-18-abi-20-/), not even for Tegra TX1.

RaumZeit, 2016-04-23:
Fixed in https://github.com/RaumZeit/PKGBUILDs/commit/a2c3fb637892f0c8782ed1cc612332199466af1e

closed 2016-04-23 by RaumZeit