RaumZeit/LinuxOnAcerCB5-311 #10: Installation failed: linux/compiler-gcc6.h: No such file or directory

stephenfrancis, 2016-06-07:
Hi,
Thank you for writing this project! I've realized that the locked-down Chromeos world is not for me and am longing to run a nice slim linux o/s on my Acer Chromebook.

Everything seemed to be going fine until I got to the "makepkg -si" step. Output pasted below. My machine is:
Acer Chromebook 13 CB5-311, RMN Z3ENN. 16GB flash.

Any help would be greatly appreciated!

[alarm@alarm ~]$ pwd
/home/alarm
[alarm@alarm ~]$ git clone https://github.com/RaumZeit/PKGBUILDs.git -b acer_cb_tegrak1
Cloning into 'PKGBUILDs'...
fatal: unable to access 'https://github.com/RaumZeit/PKGBUILDs.git/': Could not resolve host: github.com
[alarm@alarm ~]$ git clone https://github.com/RaumZeit/PKGBUILDs.git -b acer_cb_tegrak1
Cloning into 'PKGBUILDs'...
remote: Counting objects: 71004, done.
remote: Total 71004 (delta 0), reused 0 (delta 0), pack-reused 71004
Receiving objects: 100% (71004/71004), 36.99 MiB | 466.00 KiB/s, done.
Resolving deltas: 100% (36276/36276), done.
Checking connectivity... done.
[alarm@alarm ~]$ cd PKGBUILDs/core/linux-nyan/
[alarm@alarm linux-nyan]$ makepkg -si
==> Making package: linux-nyan 3.10.18-5 (Tue Jun  7 19:30:37 UTC 2016)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Installing missing dependencies...

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for alarm: 
resolving dependencies...
looking for conflicting packages...

Packages (7) docbook-xml-4.5-6  libxslt-1.1.29-1  bc-1.06.95-2  docbook-xsl-1.79.1-1  dtc-1.4.1-1  uboot-tools-2016.03-1  xmlto-0.0.28-1

Total Download Size:    1.36 MiB
Total Installed Size:  18.68 MiB

:: Proceed with installation? [Y/n] Y
:: Retrieving packages...
 libxslt-1.1.29-1-armv7h                                                    333.1 KiB  1586K/s 00:00 [############################################################] 100%
 xmlto-0.0.28-1-armv7h                                                       23.3 KiB   583K/s 00:00 [############################################################] 100%
 docbook-xml-4.5-6-any                                                       84.0 KiB   764K/s 00:00 [############################################################] 100%
 docbook-xsl-1.79.1-1-any                                                   732.4 KiB  1786K/s 00:00 [############################################################] 100%
 bc-1.06.95-2-armv7h                                                         71.7 KiB  1792K/s 00:00 [############################################################] 100%
 dtc-1.4.1-1-armv7h                                                          55.3 KiB   790K/s 00:00 [############################################################] 100%
 uboot-tools-2016.03-1-armv7h                                                91.9 KiB  1149K/s 00:00 [############################################################] 100%
(7/7) checking keys in keyring                                                                       [############################################################] 100%
(7/7) checking package integrity                                                                     [############################################################] 100%
(7/7) loading package files                                                                          [############################################################] 100%
(7/7) checking for file conflicts                                                                    [############################################################] 100%
(7/7) checking available disk space                                                                  [############################################################] 100%
:: Processing package changes...
(1/7) installing libxslt                                                                             [############################################################] 100%
(2/7) installing xmlto                                                                               [############################################################] 100%
(3/7) installing docbook-xml                                                                         [############################################################] 100%
(4/7) installing docbook-xsl                                                                         [############################################################] 100%
(5/7) installing bc                                                                                  [############################################################] 100%
(6/7) installing uboot-tools                                                                         [############################################################] 100%
(7/7) installing dtc                                                                                 [############################################################] 100%
:: Running post-transaction hooks...
(1/2) Updating manpage index...
(2/2) Updating the info directory file...
==> Retrieving sources...
  -> Downloading ed20e6baa678327b174bc972f52408438c51e387.tar.gz...
[... 119 lines cut ...]
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
In file included from include/linux/compiler.h:54:0,
                 from include/uapi/linux/stddef.h:1,
                 from include/linux/stddef.h:4,
                 from /home/alarm/PKGBUILDs/core/linux-nyan/src/include/uapi/linux/posix_types.h:4,
                 from include/uapi/linux/types.h:13,
                 from include/linux/types.h:5,
                 from include/linux/page-flags.h:8,
                 from kernel/bounds.c:9:
include/linux/compiler-gcc.h:103:30: fatal error: linux/compiler-gcc6.h: No such file or directory
 #include gcc_header(__GNUC__)
                              ^
compilation terminated.
/home/alarm/PKGBUILDs/core/linux-nyan/src/./Kbuild:35: recipe for target 'kernel/bounds.s' failed
make[1]: *** [kernel/bounds.s] Error 1
Makefile:847: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
[alarm@alarm linux-nyan]$

RaumZeit, 2016-06-09:
The gcc6 compilation problem should be fixed latest HEAD of linux-nyan package in my PKGBUILDs repo master branch. See also https://github.com/RaumZeit/PKGBUILDs/commit/92e253fab6b0c215dead4057a0e6660e97faf5d0

closed 2016-06-09 by RaumZeit

stephenfrancis, 2016-06-09:
Hi,
Thanks for working on this issue so quickly. Unfortunately, I tried again just now and had the same problem again. Looking at the PKGBUILDs repo, it seems that your commit was to the master branch.

I'm following instructions which say:
git clone https://github.com/RaumZeit/PKGBUILDs.git -b acer_cb_tegrak1

which checks out the acer_cb_tegrak1 branch, rather than master, right?

I really appreciate your help,
Stephen

RaumZeit, 2016-06-09:
I already merged the acer_cb_tegrak1 branch into my master PKBBUILDs repo some time ago and applied most of the updates on master rather than acer_cb_tegrak1 since then. The install instructions in this repo have already been updated and I just updated the instructions at my website. Please try using the master branch instead to build linux-nyan-3.10.18-9, which I successfully compiled with gcc6 yesterday.