RaumZeit/LinuxOnAcerCB5-311 #11: Error making package: fs/cifs/netmisc.c:133:40: error: 'mapping_table_ERRHRD'

stephenfrancis, 2016-06-09:
Hi,
Really sorry, but hit another problem with "makepkg -si" stage.

Have just pulled the PKGBUILDs.git repo again, master branch, and run "makepkg -si" again. It ran well for quite a while, but then got:

....
  CC      fs/proc_namespace.o
  CC      fs/bio-integrity.o
  CC      fs/autofs4/init.o
  CC      fs/autofs4/inode.o
  CC      fs/autofs4/root.o
  CC      fs/autofs4/symlink.o
  CC      fs/autofs4/waitq.o
  CC      fs/autofs4/expire.o
  CC      fs/autofs4/dev-ioctl.o
  LD      fs/autofs4/autofs4.o
  LD      fs/autofs4/built-in.o
  LD      fs/cifs/built-in.o
  CC [M]  fs/cifs/cifsfs.o
  CC [M]  fs/cifs/cifssmb.o
  CC [M]  fs/cifs/cifs_debug.o
  CC [M]  fs/cifs/connect.o
  CC [M]  fs/cifs/dir.o
  CC [M]  fs/cifs/file.o
  CC [M]  fs/cifs/inode.o
  CC [M]  fs/cifs/link.o
  CC [M]  fs/cifs/misc.o
  CC [M]  fs/cifs/netmisc.o
fs/cifs/netmisc.c:133:40: error: 'mapping_table_ERRHRD' defined but not used [-Werror=unused-const-variable=]
 static const struct smb_to_posix_error mapping_table_ERRHRD[] = {
                                        ^~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
scripts/Makefile.build:308: recipe for target 'fs/cifs/netmisc.o' failed
make[2]: *** [fs/cifs/netmisc.o] Error 1
scripts/Makefile.build:455: recipe for target 'fs/cifs' failed
make[1]: *** [fs/cifs] Error 2
Makefile:807: recipe for target 'fs' failed
make: *** [fs] Error 2
==> ERROR: A failure occurred in build().
    Aborting...

Any help would be most welcome!

Stephen

RaumZeit, 2016-06-10:
Hi again.
For some reason, compilation of the kernel package ran through smoothly using distccd, although both my Chromebook and the compilation host are using GCC 6.1.1. But I got the same errors when I tried compiling the kernel with the Chromebook directly. All of them seem to be related to the rather excessive number of warnings the compiler generates. Even though some of the warnings may not be harmful at all, compilation is aborted upon any warnings, due to the presence of the -Werror flag.
I now explicitly disabled to treat certain warnings as errors with my latest commit to the linux-nyan PKGBUILD. This allowed me to finally compile the kernel natively using my Chromebook without any errors. Can you please report back, if this last commit also works for you?

stephenfrancis, 2016-06-11:
Hi,
Thanks for that. The compilation process works all the way through now, which is great, thanks!

Unfortunately, when I reboot, I'm back in normal Chromeos. No idea why.

Last part of the compilation process pasted below.

Thanks for your help.

==> Starting package_linux-nyan-headers()...
==> Tidying install...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Compressing man and info pages...
==> Checking for packaging issue...
==> Creating package "linux-nyan-headers"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: linux-nyan 3.10.18-9 (Sat Jun 11 10:04:49 UTC 2016)
==> Installing linux-nyan package group with pacman -U...
[sudo] password for alarm: 
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (2) linux-nyan-3.10.18-9  linux-nyan-headers-3.10.18-9

Total Installed Size:  171.43 MiB
Net Upgrade Size:       13.30 MiB

:: Proceed with installation? [Y/n] y
(2/2) checking keys in keyring                                                                       [############################################################] 100%
(2/2) checking package integrity                                                                     [############################################################] 100%
(2/2) loading package files                                                                          [############################################################] 100%
(2/2) checking for file conflicts                                                                    [############################################################] 100%
(2/2) checking available disk space                                                                  [############################################################] 100%
:: Processing package changes...
(1/2) upgrading linux-nyan                                                                           [############################################################] 100%
>>> Updating module dependencies. Please wait ...
A new kernel version needs to be flashed onto /dev/mmcblk0p6.
Do you want to do this now? [y|N]
y
10272+0 records in
10272+0 records out
5259264 bytes (5.3 MB, 5.0 MiB) copied, 1.29167 s, 4.1 MB/s
(2/2) upgrading linux-nyan-headers                                                                   [############################################################] 100%
[alarm@alarm linux-nyan]$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       8.8G  5.7G  2.6G  69% /
devtmpfs        991M     0  991M   0% /dev
tmpfs           991M  105M  887M  11% /dev/shm
tmpfs           991M  480K  991M   1% /run
tmpfs           991M     0  991M   0% /sys/fs/cgroup
tmpfs           991M   76K  991M   1% /tmp
tmpfs           199M   12K  199M   1% /run/user/1000
[alarm@alarm linux-nyan]$ ls /dev/mmc*
/dev/mmcblk0       /dev/mmcblk0boot1  /dev/mmcblk0p10  /dev/mmcblk0p12  /dev/mmcblk0p3  /dev/mmcblk0p5  /dev/mmcblk0p7  /dev/mmcblk0p9
/dev/mmcblk0boot0  /dev/mmcblk0p1     /dev/mmcblk0p11  /dev/mmcblk0p2   /dev/mmcblk0p4  /dev/mmcblk0p6  /dev/mmcblk0p8  /dev/mmcblk0rpmb
[alarm@alarm linux-nyan]$ ls /dev/mmcblk0p6 
/dev/mmcblk0p6

closed 2016-06-11 by stephenfrancis