RaumZeit/LinuxOnAcerCB5-311 #12: Can not connect to WiFi

lovedheart, 2016-06-24:
Hi,

I just install with the sh script. It can boot into archlinux, but I can not connect it to router

RaumZeit, 2016-06-26:
What exactly is not working? Can you elaborate a little what your actual problem is?

lovedheart, 2016-06-27:
I choose one of the network list and click it. But nothing happens. No dialog is shown to require PSK

RaumZeit, 2016-06-28:
This sounds like gnome-keyring is not running, try installing it along with seahorse to manage your keys and passwords. Make sure gnome-keyring gets started whenever you login to XFCE4: Enable ```Settings/"Session and Startup"/Advanced/"Launch GNOME Services on startup"```
and all Gnome Keyring related services in
```Settings/"Session and Startup/"Application Autostart"```. As far as I remember, the user also needs to be in the `network` group to get Network Manager working right.
See also: https://wiki.archlinux.org/index.php/NetworkManager#No_prompt_for_password_of_secured_Wi-Fi_networks

closed 2016-06-30 by RaumZeit