RaumZeit/LinuxOnAcerCB5-311 #17: Wi-Fi doesn't work on latest Arch

mastermind-, 2017-08-28:
Wi-Fi doesn't seem to work well. I can discover networks but can't connect.
**iw dev mlan0 connect AndroidAP** produces following output in dmesg:
```
[  260.790170] mwifiex_sdio mmc1:0001:1: info: trying to associate to AndroidAP and bssid   (null)
[  260.803875] mwifiex_sdio mmc1:0001:1: info: scan: num_probes = 4
[  260.896246] gk20a 57000000.gk20a: stop latency exceeded, new value 105917 ns
[  261.431712] mwifiex_sdio mmc1:0001:1: info: trying to associate	to 'AndroidAP' bssid ac:37:43:4c:9e:87
[  262.678020] mwifiex_sdio mmc1:0001:1: info: associated to bssid ac:37:43:4c:9e:87 successfully
[  262.679034] mwifiex_sdio mmc1:0001:1: disconnecting from ac:37:43:4c:9e:87
[  262.683738] mwifiex_sdio mmc1:0001:1: info: successfully disconnected from ac:37:43:4c:9e:87:	reason code 3
[  262.683800] mwifiex_sdio mmc1:0001:1: info: successfully disconnected from 00:00:00:00:00:00: code 3
[  262.687273] cfg80211: Calling CRDA to update world regulatory domain
[  262.794593] mwifiex_sdio mmc1:0001:1: info: scan: num_probes = 4
```
AndroidAP is open network.
Kernel version is 3.10.18-21.
I used install script without modifications. Am I missing something? Can you tell me last working kernel version?

RaumZeit, 2017-09-03:
Haven't had any problems with my chromebook and WIFI connections so far. But I just pushed an update to the kernel to 3.10.18-23, maybe you try that and see whether it changes anything for you.

mastermind-, 2017-09-27:
It's working now with the latest kernel, closing the issue.

closed 2017-09-27 by mastermind-

mathuin, 2017-12-08:
I am getting this exact issue.  The kernel I have is 3.10.18-24-ARCH, were there any regressions?

mathuin, 2017-12-08:
I have isolated the problem.  I have disabled `wpa_supplicant.service` and enabled `wpa_supplicant@mlan0.service`, but when I reboot, `wpa_supplicant.service` starts and the interface-specific one does not.  I have no idea why.