RaumZeit/LinuxOnAcerCB5-311 #18: fresh install xorg not working

zombah, 2017-11-27:
Currently after installation lightdm wont start error no /usr/bin/X and gpu-nvidia-tegra-k1-x11 package not installed. If i try to installl it manually it says X-ABI-VIDEODRV_VERSION=19 dependency not found, probably some problem happen in xorg downgrade?

mathuin, 2017-12-08:
I reran the downgrade script by hand and noticed that `libxfont` is a dependency of `xorg-server` and `xorg-server-xvfb`. I could not find `libxfont` out there, but I *could* find `libxfont2`.

The right way to solve this would probably be for @RaumZeit to build `libxfont-1.5.1-1` and post it in their repository.

RaumZeit, 2017-12-09:
Thanks for the bug report...
Everything should be working again since commit 50b286b6ddb6a17cad6efd0e2be1e658a69013aa
_USB install with that commit works fine for me..._

I've also added the PKGBUILDs for xorg downgrade to this repo to enable others to build the required packages themselves

closed 2017-12-09 by RaumZeit