RaumZeit/LinuxOnAcerCB5-311 #20: Post-Install Hang On Boot [again]

KenDickey, 2017-12-09:
Thanks much for the fast response.

I get pretty much the same behavior as before.  The wifi keeps trying to connect (info scan; mlan0 not ready) and periodically I get "Framebuffer console (de)activated"..  No keyboard response aside from the power key.

Note that I did this via:
VVVV
wget https://raw.githubusercontent.com/RaumZeit/LinuxOnAcerCB5-311/archlinux/archlinux.sh
sudo bash archlinux.sh
^^^^

I did this once, as the SD partitions were OK.  The downloads looked fine.
 
Is there anything I can try/look-at here which would help?

Thanks much,
-KenD

RaumZeit, 2017-12-10:
That's odd.
The error you've been describing earlier pretty much sounded like failed nvidia driver/xorg install. That's what should be fixed now. I've tested the script yesterday by making a USB install and everything seemed fine.
Can you upload the `archlinux-install.log` somewhere (it's located in the ChromeOS directory you initiated the install from), so I can have a look at what might be going wrong during install?

P.S.: Regarding the WiFi, as far as I remember I've never used netctl on the Chromebook. I do remember having some WiFi problems initially, though, but never had any problems since using NetworkManager...

zombah, 2017-12-10:
I made install to SDCard today it works fine, at least xorg working and was able to enter xfce.
Connection to wpa2 working with wifi-menu fine.
NM probably need gnome-keyring to work.
Thanks @RaumZeit

KenDickey, 2017-12-10:
The good news is that I get the same behavior booting from the SD card. (reproducible) 

The bad news is that the boot hangs in the same place.
[archlinux-install.log](https://github.com/RaumZeit/LinuxOnAcerCB5-311/files/1546081/archlinux-install.log)

RaumZeit, 2017-12-11:
According to the `archlinux-install.log` you've posted, you are still using the old `archlinux.sh` install script. 
As I suspected, the xorg downgrade fails due to missing `libxfont` package and installation of the proprietarty NVIDIA drivers fails subsequently. As for the unresponsive keyboard you have, I think this might be a side-effect of `lightdm` trying to start the xserver over and over again but crashing each time...

Please try installation with a freshly downloaded `archlinux.sh` which fixes the above bug...

@zombah I'll see that the `gnome-keyring` package will be included in the next version of the script, to prevent `nm` troubles, thanks!

KenDickey, 2017-12-11:
Thanks again!  Must be my error.  WGET returned a script which differed and I am now replying from my Acer CB5-311 via Arch Linux.

It boots !!!!   8^)  8^O  8^}

THANK YOU THANK YOU THANK YOU!!

closed 2017-12-26 by RaumZeit