RaumZeit/LinuxOnAcerCB5-311 #21: Checksum problem w PKGBUILDs/core/linux-nyan

KenDickey, 2017-12-12:
Greetings,

I am following script in 
  https://www.tbi.univie.ac.at/~ronny/acer-cb5-311.html#setupou help me out?

but getting checksum failures.

Can you help me out?

Thanks a bunch,
-KenD
VVV=========================VVV
git clone https://github.com/RaumZeit/PKGBUILDs.git
cd PKGBUILDs/core/linux-nyan
makepkg -si
^^^
[invalid-mkpkg.txt](https://github.com/RaumZeit/LinuxOnAcerCB5-311/files/1553370/invalid-mkpkg.txt)
=========================^^^

KenDickey, 2017-12-23:
Work around:

I just substituted recomputed checksums on the 2 files into PKGBUILDs/core/linux-nyan/PKGBUILD

Build and install went fine.  I have no idea why the checksums failed to match, so do not know if this was the right thing to do.

FYI,
-KenD

RaumZeit, 2018-02-25:
I know this problem. Sometime, google decides to play around with the git repository. This means, that the IDs of particular commits and tags magically move ;) Consequently, the packaged snapshots do not match the checksums anymore and updating the checksum in the PKGBUILD is the only option to circumvent this problem...

closed 2018-02-25 by RaumZeit