RaumZeit/LinuxOnAcerCB5-311 #22: error while loading shared libraries: libcryptsetup.so.12: cannot open shared object file: No such file or directory

smok874, 2017-12-15:
After done of installing a Xorg basics I take message 
'umount {..}/mmcblk0p7 is busy
(In some cases useful info about processes that use
         the device is found by lsof(8) or fuser(1))'

in logs i saw a 'libcryptsetup.so.12' is the problem

[archlinux-install_.log](https://github.com/RaumZeit/LinuxOnAcerCB5-311/files/1562611/archlinux-install_.log)

jorants, 2017-12-26:
A few weeks ago libcryptsetup was updated, but the tarball that is grabed for the base install is outdated and contains the old version.
Doing a pacman -Syyu instead of pacman -Syy in the first pacman call should update to the new version of cryptsetup.
This seems like a good idea anyway since the tarball might be behind the repo.
you can edit the script by hand or run the attached patch. (i will make a pull request as well)

If you want you can also at cryptsetup to the list of core packages, instead of updating everything.
[cryptsetup.patch.txt](https://github.com/RaumZeit/LinuxOnAcerCB5-311/files/1587306/cryptsetup.patch.txt)

smok874, 2017-12-26:
Thank you it works very well

closed 2017-12-26 by smok874