RaumZeit/LinuxOnAcerCB5-311 #24: Wireless driver fail with existing fix

mathuin, 2018-01-01:
Lately I have been suffering network failures after running for ten minutes or so.

The log entries look like this:
```
[  663.206464] mwifiex_sdio mmc1:0001:1: GET_CMD_NODE: cmd node not available
[  663.206474] mwifiex_sdio mmc1:0001:1: PREP_CMD: no free cmd node
```
Google shows that this issue arose in 2015, and was fixed with this patch:  https://www.spinics.net/lists/linux-wireless/msg137134.html

Is it possible to get a patched version of the wireless driver for this Chromebook?

RaumZeit, 2018-02-25:
I don't see any of such problems... maybe there has been an issue with the kernel you are using?

mathuin, 2018-02-25:
Could have been.  I had to revert to ChromeOS to continue to use the system, so I'm not sure.   You're welcome to close the issue if you think it's not likely to be reproduced elsewhere.

closed 2019-04-10 by RaumZeit