RaumZeit/LinuxOnAcerCB5-311 #25: Running mainline

reey, 2018-01-10:
@RaumZeit 
I have forked your project and edited the script to install mainline kernel instead of the chromeOS kernel.
Installing works fine, but Wifi is not working..
Maybe you could give it a try, as I am not the experienced..

reey

reey, 2018-02-15:
seems like wifi is working now, with kernel 4.15

RaumZeit, 2018-02-25:
Thanks for the suggestion! I'll have a look at it as soon as I find some time.
For the time being, the provided install script was mainly for providing the means to install the proprietary NVIDIA GPU drivers. This is not possible with mainline linux, though.
But let's see how the upgrades to `mesa`, `nouveau`, and mainline kernel develop, maybe using the official NVIDIA drivers becomes obsolte at some point...

closed 2018-02-25 by RaumZeit