RaumZeit/LinuxOnAcerCB5-311 #26: Critical BUG  Repository compromise because Xorg-Server 1.17 LIBXFONT GONE

aperfeito, 2018-02-15:
System Arch Linux
Linux localhost 3.10.18 #1 SMP Mon Jan 8 23:06:44 PST 2018 armv7l GNU/Linux
Upgrade via pacman in 02-11-2018
Xorg-sever version : 1.17

After upgrade ARCHLINUX ARM I receive this suprise:
usr/lib/xorg-server/Xorg: error while loading shared libraries: libXfont.so.1: cannot open shared object file: No such file or directory

Because the package **libxfont** MUST BE REMOVE FOR CONTINUE UPGRADE THIS OS

After this Lightdm.service FAILED and don't start lightdm proprely

HELP !!! :|

Reproduce the error:

sudo pacman -Syyuu
:: Synchronizing package databases...
core 127.8 KiB 2.60M/s 00:00 [####################################################] 100%
extra 1594.6 KiB 9.55M/s 00:00 [####################################################] 100%
community 4.1 MiB 7.79M/s 00:01 [####################################################] 100%
multilib 166.6 KiB 12.5M/s 00:00 [####################################################] 100%
archlinuxfr 12.1 KiB 11.8M/s 00:00 [####################################################] 100%
:: Starting full system upgrade...
:: Replace compositeproto with extra/xorgproto? [Y/n] y
:: Replace damageproto with extra/xorgproto? [Y/n] y
:: Replace dmxproto with extra/xorgproto? [Y/n] y
:: Replace fixesproto with extra/xorgproto? [Y/n] y
:: Replace fontsproto with extra/xorgproto? [Y/n] y
:: Replace inputproto with extra/xorgproto? [Y/n] y
:: Replace kbproto with extra/xorgproto? [Y/n] y
:: Replace randrproto with extra/xorgproto? [Y/n] 
:: Replace recordproto with extra/xorgproto? [Y/n] y
:: Replace renderproto with extra/xorgproto? [Y/n] y
:: Replace scrnsaverproto with extra/xorgproto? [Y/n] y
:: Replace videoproto with extra/xorgproto? [Y/n] y
:: Replace xextproto with extra/xorgproto? [Y/n] y
:: Replace xf86dgaproto with extra/xorgproto? [Y/n] y
:: Replace xf86vidmodeproto with extra/xorgproto? [Y/n] y
:: Replace xineramaproto with extra/xorgproto? [Y/n] y
:: Replace xproto with extra/xorgproto? [Y/n] y
resolving dependencies...
looking for conflicting packages...
error: failed to prepare transaction (could not satisfy dependencies)
:: libxfont: removing fontsproto breaks dependency 'fontsproto>=2.1.3'

After your remove libxfont necessary to xorg-server version 17.1 with

pacman -Rdd libxfont

The upgrade goes OK, After you reboot the System the ArchLiniux OS is faled to start lightdm.service and the error usr/lib/xorg-server/Xorg: error while loading shared libraries: libXfont.so.1: cannot open shared object file: No such file or directory

HELP  :|

Try to RollBAck
Give THIS
sudo pacman -U libxfont-1.5.3-1-armv7h.pkg.tar.xz
loading packages...
resolving dependencies...
looking for conflicting packages...
:: fontsproto and xorgproto are in conflict. Remove xorgproto? [y/N] Y
fontsproto and xorgproto are in conflict. Remove xorgproto? [y/N] Y
error: failed to prepare transaction (could not satisfy dependencies)
:: libxfont: requires xproto
:: libdmx: removing xorgproto breaks dependency 'dmxproto'
:: libice: removing xorgproto breaks dependency 'xproto'
:: libx11: removing xorgproto breaks dependency 'xproto'
:: libx11: removing xorgproto breaks dependency 'kbproto'
:: libxau: removing xorgproto breaks dependency 'xproto'
:: libxcomposite: removing xorgproto breaks dependency 'compositeproto'
:: libxdamage: removing xorgproto breaks dependency 'damageproto'
:: libxdmcp: removing xorgproto breaks dependency 'xproto'
:: libxext: removing xorgproto breaks dependency 'xextproto'
:: libxfixes: removing xorgproto breaks dependency 'fixesproto'
:: libxfont2: removing xorgproto breaks dependency 'xproto'
:: libxi: removing xorgproto breaks dependency 'inputproto'
:: libxinerama: removing xorgproto breaks dependency 'xineramaproto'
:: libxrandr: removing xorgproto breaks dependency 'randrproto'
:: libxrender: removing xorgproto breaks dependency 'renderproto'
:: libxss: removing xorgproto breaks dependency 'scrnsaverproto'
:: libxtst: removing xorgproto breaks dependency 'recordproto'
:: libxtst: removing xorgproto breaks dependency 'inputproto'
:: libxv: removing xorgproto breaks dependency 'videoproto'
:: libxxf86dga: removing xorgproto breaks dependency 'xf86dgaproto'
:: libxxf86vm: removing xorgproto breaks dependency 'xf86vidmodeproto'

Hey Guys look this

https://bugs.archlinux.org/task/57499

https://bugs.archlinux.org/task/57494

We have to do something ???

RaumZeit, 2018-02-19:
I am aware of this and working on a fix...

P.S.: Don't report bugs on ArchLinuxARM to ArchLinux. These two projects are separate from each other, and usually the maintainers for x86 ArchLinux don't like being spammed with ARM problems, especially when it comes to old PKGBUILDs

aperfeito, 2018-02-19:
Hi, RaumZeit,

    About the bugs report you mention now I know.
    God Bless you man. Everything here the status is stop.All because this bug
     Wishes you find the solution quick about this.  

Best Regards
Perfeito, Alexander

RaumZeit, 2018-02-25:
Problem solved:
Either do a re-install of the system using the latest installer script, or install the following package:
```
curl -L -O https://www.tbi.univie.ac.at/~ronny/alarm/libxfont-1.5.3-1.1-armv7h.pkg.tar.xz
sudo pacman -U libxfont-1.5.3-1.1-armv7h.pkg.tar.xz
```
After that, upgrades should run smoothly, as this package actually allows for the transition from `fontsproto` to `xorgproto` (See also https://github.com/RaumZeit/LinuxOnAcerCB5-311/commit/725bdbd48c90c4cc9ce4361ae508dda143bb543f)

closed 2018-02-25 by RaumZeit

aperfeito, 2018-02-25:
Raum,
     That's ok now !! new package libxfont-1.5.3-1.1 (for Nvidia Systems ARM using version x-sever 1.17)
God Bless you !!!
Thanks

Best Regards

Perfeito, Alexandre