RaumZeit/LinuxOnAcerCB5-311 #27: wget removed in chromeos > 64, dev_install leads to error 403

Spleeeno, 2018-02-18:
hey there,

so I tried running the script on chromeos 65 and it turns out they removed wget.
so my google-fu lead me to: https://github.com/headmelted/codebuilds/issues/40
main problem is: the dev_install line leads to error 403, because - I guess - they havent packaged the tools for nyan_big yet.
so, if anyone's got any tips or tricks - that would be appreciated ;)

reey, 2018-02-18:
You could try chromebrew to install wget

Spleeeno, 2018-02-23:
had to ctrl+alt+f2, login with the root-pw I setup during the dev-mode installation and do a passwd chronos cause the sudo didnt work with blank/password/chrome/etc.
but its installing right now, so there's that :+1:
btw, will [this](https://www.phoronix.com/scan.php?page=news_item&px=NVIDIA-Tegra-Mesa-2018) make it possible to run stock arm kernel and noveau in the future? maybe even wayland?

RaumZeit, 2018-02-25:
Missing `wget` is not resolved as the latest install script uses `curl` for downloads...

As for the mainline kernel and changes in mesa, I'll have a look at it. Thanks for the hint.

closed 2018-02-25 by RaumZeit