RaumZeit/LinuxOnAcerCB5-311 #29: umount: /tmp/arfs/dev: target is busy

Manawyrm, 2018-09-04:
Hi, 

I'm trying to install Arch on this CB5-311 with your script. 
It's a fresh installation of ChromeOS, without any software/plugins installed. 
After the second run of the script, this happens: 

```
WARNING! This script will install binary packages from an unofficial source!

If you don't trust me (Ronny Lorenz a.k.a. RaumZeit) press CTRL+C to quit
(see https://github.com/RaumZeit/LinuxOnAcerCB5-311 for further details)

Press [Enter] to proceed installation of ArchLinuxARM

Chrome device model is: BIG E2G-G4E-B3B-A77

Installing ArchLinuxARM latest

Kernel Arch is: armv7l  Installing ArchLinuxARM Arch: armv7

Press [Enter] to continue...
Target Kernel Partition: /dev/mmcblk0p6  Target Root FS: /dev/mmcblk0p7
mke2fs 1.44.1 (24-Mar-2018)
Suggestion: Use Linux kernel >= 3.18 for improved stability of the metadata and journal checksum features.
Discarding device blocks: done                            
Creating filesystem with 5242880 4k blocks and 1310720 inodes
Filesystem UUID: bad1cc8a-ed8e-451b-96e8-6c7ae65d97c4
Superblock backups stored on blocks: 
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
        4096000

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information: done   

Downloading and extracting ArchLinuxARM rootfs   / ...done.
Copying files from ChromeOS to ArchLinuxARM rootdir   | ...done.
Installing development base packages   | ...done.
umount: /tmp/arfs/dev: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
chronos@localhost ~/Downloads $ 
``` 

Do you have any ideas on how to troubleshoot this? 

Thanks,
Tobias

Pennykoin, 2019-03-09:
Same issue 3-4 times now, giving up on chrubuntu until solved.

RaumZeit, 2019-04-10:
Sorry for the late reply...

The install script has been updated now.
The main reason it failed was an uninitialized pacman keyring, but even after fixing that, slow downloads would have also killed the installation procedure. The script now uses curl with a 60sec timeout for downloading packages, and any (premature) exit from the installer script tries to unmount the entire chroot correctly.

I just tried the installation multiple times and everything seems to be working as expected.

closed 2019-04-13 by RaumZeit