RaumZeit/LinuxOnAcerCB5-311 #33: Problem in script execution

ehadavi, 2021-01-20:
The script runs smoothly, but finally it gives an error for line 71 saying process(#) doesn't exist and exits. The process number (#) changes each time.

velocity37, 2022-02-06:
On the last update for these devices, the kill command doesn't seem to be available as the chronos user. If you sudo su to switch to root then the script will execute. But there's another issue #34

HAHAHA44, 2023-01-18:
> [archlinux-install.log](https://github.com/RaumZeit/LinuxOnAcerCB5-311/files/5851577/archlinux-install.log) The script runs smoothly the first time and creates the partition, but in the second run after reboot it gives an error for line 71 saying 'kill: (#) No such process' and exits (The process number (#) changes each time). maybe connected to changes in chrome os builds?

Hello, have you finished the problem? I faced the same thing and I use Acer C810 to install the Linux.

The kill command is available but the prcess id of gpg-agent is changing, thus the kill command failed.

The script failed with `Error: there is no secret key available to sign with`, before it, there are logs shown:

`gpg: agent_genkey failed: End of file`
`gpg: key generation failed: End of file`

And I searched the problem on Google, there are some relevant answer: https://stackoverflow.com/questions/72464265/gpg-key-generation-failed-end-of-file.

Maybe it is because the uncompatible gpg version? I have not do the further work.