RaumZeit/LinuxOnAcerCB5-311 #34: failed to start light display manager

jason0505, 2021-02-20:
i install the archlinux on acer chromebook 13 C810, it's also tegra K1 CPU too. everything is done but failed to start light display manager after reboot. what should i do next to solve this problem?please

velocity37, 2022-02-06:
Running into this myself. The problem can be either #33 or if you run the script as root, there's a problem with the script atm.
install_xbase fails to install downgraded xorg/libxfont because the dependencies `xproto` and `fontsproto` are no longer hosted by Arch. So the install fails with unresolved dependencies and your Arch has no X.

Trying to see if grabbing the old packages archived on archive.org at least gets something up and running. But it might cause a cascading effect relying on EOL packages.

guttenberger, 2022-04-03:
@velocity37 Thank you very much for your research, because of you I was able to get xfce4 / IceWM to work.
However for some reason WebGL is not working and Chromium is very slow.

I think i didnt found the tegra.conf you mentioned. 
Can you maybe share tegra.conf ?

velocity37, 2022-04-08:
> I think i didnt found the tegra.conf you mentioned. Can you maybe share tegra.conf ?
/usr/share/X11/xorg.conf.d/tegra.conf

Looking at gpu-nvidia-tegra-k1-x11-21.7.0-1-armv7h.pkg.tar.xz it does look like it has driver set to "nvidia" so I misattributed the conf to that package. Not sure where the tegra.conf I got with it set to "tegra" was. The file in there is correct.