RaumZeit/LinuxOnAcerCB5-311 #35: PKGBUILD step fails; Checksums have changed

kybr, 2021-07-30:
Thank you so much for this work. I have one of these chromebooks and I really want to use Arch on it. If you can look at this, I would be most appreciative!

~

Building the new kernel with `makepkg -si` fails because the checksums are not correct. Ignoring the checksums with `--skipchecksums` leads to a kernel build failure. Here's an excerpt of your PKGBUILD file from a commit a couple of years ago:

```
...
_commit=8d57cc314fb6199846e5b30637aa0ed355a44704
_coreboot_commit=24cf824297f8a4bf5faf6711fbe296d8384bddc3
...
source=(  "https://chromium.googlesource.com/chromiumos/third_party/kernel/+archive/${_commit}.tar.gz"
          "https://chromium.googlesource.com/chromiumos/third_party/coreboot/+archive/${_coreboot_commit}.tar.gz"
...
sha256sums=('b32a477312ce3db1c289ad1e5c0456ef8553468be07854ef61a16870e502ca8e'
62
            '68f9050651c9714ce99c617a008e39266fb9887e6038dba8a7cfca223e01b989'
```

Those commits don't match the checksums. I don't currently know enough about google's git or ChromeOS to fix this myself. If you could take a look.. Thanks.

kybr, 2021-07-30:
I see that there are a couple closed issues regarding checksums not matching. I will skip the checksums and focus on the kernel build error instead.

kybr, 2021-07-30:
It seems that the latest ArchLinuxARM uses gcc10 and that does not build this old kernel. I think this old kernel is patched for 5 and 6, but not 10. The simplest solution might be to use an older "latest" ArchLinuxARM image from here:

    http://tardis.tiny-vps.com/aarm/repos/

Alternatively, I could install an older gcc; I tried that with no luck.

velocity37, 2022-02-07:
You don't need to build the kernel yourself if you don't want to. The script installs a precompiled one for you as part of the install process. If you only want the kernel and not the whole OS install then you can just pinch the kernel and kernel header tarballs from the URLs in the script.